sumotoy/RA8875 #143: Image rotation in RA8875

mikey06355, 2019-08-12:
Using the example load24bitImages if I set rotation to a 1 or 3 (portrait mode) the image will not display. I am using an Adafruit 5" tft and a Teensy 3.2. Text rotate fine.
Thanks...Mike

KurtE, 2019-09-03:
For what it is worth, I found that I had the same issue.   Simply trying to draw an image I had stored in an array.  

The issue is with the function drawPixels, that does not work properly in portrait mode.  I believe this is because the underlying stuff is simply swapping the usage of what is horizontal and what is vertical... 

I talk more about it in the Teensy forum thread: https://forum.pjrc.com/threads/57280-RA8875-from-Buydisplay?p=214506&viewfull=1#post214506

KurtE, 2020-02-01:
@mikey06355 and @sumotoy  (if you are at listening)...

A couple of us have been doing some work to make this driver work on the newer Teensy boards, in particular T4.  We have a good working version up at:  https://github.com/mjs513/RA8875/tree/RA8875_t4

One of the last things I added in our version is a function that is in some of our other drivers. 
```
writeRect(int16_t x, int16_t y, int16_t w, int16_t h, const uint16_t *pcolors)
```
This function fills a rectangle on the screen with the bitmap data pointed to by pcolors.  If in portrait mode, it will call drawPixels with a temporary buffer, where it will grab the pixels from the opposite direction... in order to get it right on the display...   

More details up in the thread: https://forum.pjrc.com/threads/57280-RA8875-from-Buydisplay?p=228071&viewfull=1#post228071